stm32-rs/synopsys-usb-otg #1: Publish on crates.io

TeXitoi, 2019-12-30:
Hi!

To let you know, I've ported my keyboard using this crate successfully: https://github.com/TeXitoi/keyberon-f4

Let me know if you need any help to publish this crate.

Disasm, 2019-12-30:
Wow, nice to hear! As for the release, I'd like to have `stm32ral` released first. I already filed a PR for this: https://github.com/adamgreig/stm32ral/pull/12

Additionally, `stm32f4xx-hal` lacks proper PLL setup, without it that "f4 pill" will not work due to its 25MHz external crystal. However, since I removed `stm32f4xx-hal` from the dependency list, I can publish without this requirement.

TeXitoi, 2020-02-11:
Hi! I now have a keyboard using this driver! (same address)

Any news on publishing? the stm32ral PR should be now OK.

Thanks for the driver :-)

adamgreig, 2020-02-11:
Sadly while stm32ral 0.3.0 has been created, I can't publish it to crates.io because it's over the crate file-size limit. I've emailed crates.io to ask if we can increase it, and if not, we'll have to think of something to make the crate size a bit smaller. But I guess that's blocking releasing this.

TeXitoi, 2020-02-11:
@adamgreig thanks for the feedback!

adamgreig, 2020-02-12:
stm32ral v0.3.1 is now on crates.io :tada:

Disasm, 2020-02-12:
@adamgreig thank you!

Disasm, 2020-02-12:
Published [v0.1.0](https://crates.io/crates/synopsys-usb-otg) :tada:

closed 2020-02-12 by Disasm

TeXitoi, 2020-02-14:
Thanks!

@Disasm now I need stm32f4xx-hal integration, do you want I take care of it using your fork, or do you want to do it?

Disasm, 2020-02-14:
@TeXitoi I can integrate the driver into stm32f4xx-hal in a day or two. If you have time to help with the clock setup part, please do it.

TeXitoi, 2020-02-14:
do you have any pointers on what you have already looked on the clock setup? I'll try to look at is this weekend.

Disasm, 2020-02-14:
The patch from my fork works, but the upstream repo has changed since then and now this patch can't be ported easily. Specifically, I'm talking about this change: https://github.com/stm32-rs/stm32f4xx-hal/commit/44ec5ab41437bdfc509bf944867b5e438d747963 It's difficult (and sometimes impossible) to have precise both sysclk and PLL48 frequencies, so I think that PLL48 frequency should be more "important" when it's requested by user.

TeXitoi, 2020-02-14:
OK, I'll try to look at it.