stm32-rs/synopsys-usb-otg #10: Support low-powered devices

engstad, 2020-09-18:
I went through the Reference Manual for the low-power devices STM32L4{7,8,9,A}X, and it seems like there are some differences in the OTG HW:
* These have 6 end-points, not 4.
* They have additional global control registers for low-power (`GLPMCFG`, `GPWRON`, `GADPCTL`, `DIEPTXF{4,5}`.
* Quite a few reset-values are different (`GOTGCTL=0x0001_0000`, `GUSBCFG=0x0000_1440`, `GRSTCTL=0x8000_000`,`DIEPTXF0=0x0200_0200`, `HPTXFSIZ=0x0200_0400`, and the `DIEPTXFx` are reset to `0x0200_0200, 0x0200_0400, 0x0200_0600`, etc.

A few questions:
* I noticed that you hard-code register blocks, such as `otg_fs_device_v1::RegisterBlock` according to the layout of the F4 devices. Why is that, can you not use the SVD information?
* Would it be better to add these as `otg_fs_device_v2`, or should I just make yet another USB fork?

Disasm, 2020-09-19:
> there are some differences in the OTG HW:

Differences compared to what?

Right now, the library supports up to nine endpoints (STM32F72x OTG HS has nine), we dot use `DIEPTXF{4,5,...}` registers for this. Instead, we use a `DIEPTXFx` register with a calculated base address.

Reset values indeed can pose a problem, we should replace `write_reg!()` with `modify_reg!()` where possible. `write_reg!()` uses reset values for a specific chip, but `modify_reg!()` doesn't.

As for hard-coding register blocks: we used `stm32ral` definitions earlier, but then we faced with the problem when you want to use `stm32ral` in a main project for a different chip. We used register definitions for a specific chip, and it was impossible to enable two different chip features in this library. That's why we decided to pull all the necessary register definitions into the USB driver, like we did for `stm32-usbd`.

I don't think we need to add another set of register definitions, but if you know a specific chip with better and more complete register definitions, we can replace the ones we have now.

BTW, what problem are you trying to solve? I hope that this library should just work with L4 chips, despite not being supported out of the box in `stm32l4xx-hal`.

engstad, 2020-09-21:
Thanks for the reply.

I have an STM32L486RX board that I made myself that I want to use the USB for. However, when I tried using your library, it doesn't work. Therefore, I looked into it more deeply, and that's where I noticed the differences w.r.t. your code and the Reference Manual. I'm no expert on USB, but I think that it doesn't work because some of the new registers (w.r.t. to low-power operation) are not set up correctly.

If I get the time, I might mess with your code to see if I can get it working.

Leo1003, 2020-12-12:
Hello, I tested the driver on my STM32L476RG board today. It can work successfully with a USB serial example.

Initially, it stucked in enabling the USB device as @engstad 's did.
After some investigation, I found out that the clock for USB is not set because of other board can easily use hsi48 as their USB clock source. However, L476 and L486 doesn't have hsi48, they can only use PLLQ, PLLSAI1Q, or MSI clock as USB 48MHz clock source. 

Currently, their is no way to set those registers by using the HAL crates. So, you have to modify the registers directly.

Following is my codes for setting up the clock: (attach to PLLQ)
```rust
fn enable_pllq_48mhz() {
    let rcc = unsafe { &(*stm32::RCC::ptr()) };
    // PllQ = PllDivider::Div2
    rcc.pllcfgr.modify(|_, w| unsafe { w.pllq().bits(0x00) });
    // Enable PllQ
    rcc.pllcfgr.modify(|_, w| w.pllqen().set_bit());
    // Attach 48MHz clock source onto PllQ
    rcc.ccipr.modify(|_, w| unsafe { w.clk48sel().bits(0x02) });
}

fn main() -> ! {
    let dp = stm32::Peripherals::take().unwrap();

    let mut flash = dp.FLASH.constrain();
    let mut rcc = dp.RCC.constrain();
    let mut pwr = dp.PWR.constrain(&mut rcc.apb1r1);

    let clocks = rcc
        .cfgr
        .pll_source(PllSource::HSI16)
        .sysclk_with_pll(48.mhz(), PllConfig::new(2, 12, PllDivider::Div2))
        .hclk(48.mhz())
        .pclk1(24.mhz())
        .pclk2(24.mhz())
        .freeze(&mut flash.acr, &mut pwr);

    enable_pllq_48mhz();

    /* USB serial setup... */
    loop { /* USB serial loop... */ }
}
```

engstad, 2021-04-21:
I also managed to get it to work. Please see: https://github.com/stm32-rs/stm32l4xx-hal/pull/216

closed 2021-04-21 by engstad