stm32-rs/synopsys-usb-otg #13: Interrupts don't work

Disasm, 2021-01-10:
It was reported that OTG_HS interrupts never fired on F429 disco configured with "HS in FS" mode.

TomDeRybel, 2021-06-22:
I'm also having issues with OTG2_HS not generating any interrupts.
The platform is an ST Nucleo-H743ZI2 (chip HAL identifier: stm32h743v). The USB port is on OTG2.

To reproduce the issue, I have made two examples: 

- One using blocking (polling) which works
- One using interrupts in RTIC which fails to enumerate (see log output below). 
- For the interrupt case, no interrupts seem to ever occur (the RTIC task is never triggered on any of the USB interrupts).

[blinky-nucleo-h743zi2-blocking.zip](https://github.com/stm32-rs/synopsys-usb-otg/files/6693982/blinky-nucleo-h743zi2-blocking.zip)

[blinky-nucleo-h743zi2-interrupt.zip](https://github.com/stm32-rs/synopsys-usb-otg/files/6693983/blinky-nucleo-h743zi2-interrupt.zip)

Output from "tail -f /var/log/syslog"
Jun 22 13:40:53 Feynman kernel: [20339.748558] usb 1-5.3.1: new full-speed USB device number 30 using ehci-pci
Jun 22 13:41:09 Feynman kernel: [20355.487171] usb 1-5.3.1: device descriptor read/64, error -110
Jun 22 13:41:25 Feynman kernel: [20371.358790] usb 1-5.3.1: device descriptor read/64, error -110
Jun 22 13:41:25 Feynman kernel: [20371.531792] usb 1-5.3.1: new full-speed USB device number 31 using ehci-pci
Jun 22 13:41:41 Feynman kernel: [20387.231404] usb 1-5.3.1: device descriptor read/64, error -110
Jun 22 13:41:57 Feynman kernel: [20403.103016] usb 1-5.3.1: device descriptor read/64, error -110
Jun 22 13:41:57 Feynman kernel: [20403.211345] usb 1-5.3-port1: attempt power cycle
Jun 22 13:41:57 Feynman kernel: [20403.791146] usb 1-5.3.1: new full-speed USB device number 32 using ehci-pci
Jun 22 13:42:09 Feynman kernel: [20415.019728] usb 1-5.3.1: device not accepting address 32, error -110
Jun 22 13:42:09 Feynman kernel: [20415.086755] usb 1-5.3.1: new full-speed USB device number 33 using ehci-pci
Jun 22 13:42:20 Feynman kernel: [20426.282532] usb 1-5.3.1: device not accepting address 33, error -110
Jun 22 13:42:20 Feynman kernel: [20426.282695] usb 1-5.3-port1: unable to enumerate USB device
Jun 22 13:42:20 Feynman kernel: [20426.691452] usb 1-5.3.1: new full-speed USB device number 34 using ehci-pci
Jun 22 13:42:36 Feynman kernel: [20442.013112] usb 1-5.3.1: device descriptor read/64, error -110
Jun 22 13:42:51 Feynman kernel: [20457.884764] usb 1-5.3.1: device descriptor read/64, error -110
Jun 22 13:42:52 Feynman kernel: [20458.058682] usb 1-5.3.1: new full-speed USB device number 35 using ehci-pci
Jun 22 13:43:07 Feynman kernel: [20473.758346] usb 1-5.3.1: device descriptor read/64, error -110
Jun 22 13:43:23 Feynman kernel: [20489.629028] usb 1-5.3.1: device descriptor read/64, error -110
Jun 22 13:43:23 Feynman kernel: [20489.737134] usb 1-5.3-port1: attempt power cycle
Jun 22 13:43:24 Feynman kernel: [20490.316936] usb 1-5.3.1: new full-speed USB device number 36 using ehci-pci
Jun 22 13:43:35 Feynman kernel: [20501.544651] usb 1-5.3.1: device not accepting address 36, error -110
Jun 22 13:43:35 Feynman kernel: [20501.612632] usb 1-5.3.1: new full-speed USB device number 37 using ehci-pci
Jun 22 13:43:38 Feynman kernel: [20504.232701] usb 1-5.3.1: device not accepting address 37, error -32
Jun 22 13:43:38 Feynman kernel: [20504.232877] usb 1-5.3-port1: unable to enumerate USB device
Jun 22 13:43:47 Feynman kernel: [20513.736402] usb 1-5.3.1: new full-speed USB device number 38 using ehci-pci
Jun 22 13:44:03 Feynman kernel: [20529.051003] usb 1-5.3.1: device descriptor read/64, error -110
Jun 22 13:44:18 Feynman kernel: [20544.922585] usb 1-5.3.1: device descriptor read/64, error -110
Jun 22 13:44:19 Feynman kernel: [20545.095611] usb 1-5.3.1: new full-speed USB device number 39 using ehci-pci
Jun 22 13:44:34 Feynman kernel: [20560.794299] usb 1-5.3.1: device descriptor read/64, error -110
Jun 22 13:44:45 Feynman systemd[1721]: snap.code.code.75581f33-a6b9-41fc-8ec7-62f7c19a574b.scope: Succeeded.
Jun 22 13:44:50 Feynman kernel: [20576.665904] usb 1-5.3.1: device descriptor read/64, error -110
Jun 22 13:44:50 Feynman kernel: [20576.775137] usb 1-5.3-port1: attempt power cycle
Jun 22 13:44:51 Feynman kernel: [20577.354936] usb 1-5.3.1: new full-speed USB device number 40 using ehci-pci
Jun 22 13:44:53 Feynman kernel: [20579.710798] usb 1-5.3.1: device not accepting address 40, error -32

I hope this helps to diagnose he issue. I'm willing to help and run tests in case you do not have this platform.

Disasm, 2021-06-23:
Thank you for the information! Do you know if the same problem happens with OTG1_HS?

TomDeRybel, 2021-06-23:
I don't know. The Nucleo board does not have that port connected.
But, give me an hour and I'll see if I can break it out without breaking anything. BRB

TomDeRybel, 2021-06-23:
I've connected the OTG1_HS port. Now, these pins also connect to other things on the Nucleo board (such as LD3), so the USB bus is not happy at all.... I'm holding-off on cutting any traces for now.

--> However, the the interesting thing is that the OTG1_HS interrupts do indeed function, while the OTG2_HS interrupts do not.

When I connect OTG2_HS, the debug console does not show any interrupts firing. 
When I connect OTG1_HS, the debug console shows the interrupts firing.

Here is the updated firmware for your reference:

[blinky-nucleo-h743zi2-interrupt_2021-06-23.zip](https://github.com/stm32-rs/synopsys-usb-otg/files/6700957/blinky-nucleo-h743zi2-interrupt_2021-06-23.zip)

Here are the various debug outputs:

Output from "tail -f /var/log/syslog"
Jun 23 12:04:34 Feynman kernel: [10600.249633] usb 5-1.3.2: new low-speed USB device number 45 using uhci_hcd
Jun 23 12:04:34 Feynman kernel: [10600.328572] usb 5-1.3.2: device descriptor read/64, error -71
Jun 23 12:04:34 Feynman kernel: [10600.516591] usb 5-1.3.2: device descriptor read/64, error -71
Jun 23 12:04:35 Feynman kernel: [10600.689599] usb 5-1.3.2: new low-speed USB device number 46 using uhci_hcd
Jun 23 12:04:35 Feynman kernel: [10600.768641] usb 5-1.3.2: device descriptor read/64, error -71
Jun 23 12:04:35 Feynman kernel: [10600.956640] usb 5-1.3.2: device descriptor read/64, error -71
Jun 23 12:04:35 Feynman kernel: [10601.062848] usb 5-1.3-port2: attempt power cycle
Jun 23 12:04:35 Feynman kernel: [10601.651704] usb 5-1.3.2: new low-speed USB device number 47 using uhci_hcd
Jun 23 12:04:36 Feynman kernel: [10602.069584] usb 5-1.3.2: device not accepting address 47, error -71
Jun 23 12:04:36 Feynman kernel: [10602.137580] usb 5-1.3.2: new low-speed USB device number 48 using uhci_hcd
Jun 23 12:04:36 Feynman kernel: [10602.557636] usb 5-1.3.2: device not accepting address 48, error -71
Jun 23 12:04:36 Feynman kernel: [10602.558821] usb 5-1.3-port2: unable to enumerate USB device

Output from the debug console:
0 INFO  Blinky on Nucleo-H743ZI2 - stm32h7xx-hal + RTIC example
└─ blinky_nucleo_h743zi2_interrupt::app::init @ src/main.rs:116
1 INFO  Setup PWR...
└─ blinky_nucleo_h743zi2_interrupt::app::init @ src/main.rs:119
2 INFO  Setup RCC...
└─ blinky_nucleo_h743zi2_interrupt::app::init @ src/main.rs:124
3 INFO  Setup Timers...
└─ blinky_nucleo_h743zi2_interrupt::app::init @ src/main.rs:163
4 INFO  Setup GPIO...
└─ blinky_nucleo_h743zi2_interrupt::app::init @ src/main.rs:171
5 INFO  Setup USB 2...
└─ blinky_nucleo_h743zi2_interrupt::app::init @ src/main.rs:184
6 INFO  Setup USB 1...
└─ blinky_nucleo_h743zi2_interrupt::app::init @ src/main.rs:213
7 INFO  USB interrupt received.
└─ blinky_nucleo_h743zi2_interrupt::app::usb_handler @ src/main.rs:306
8 INFO  USB interrupt received.
└─ blinky_nucleo_h743zi2_interrupt::app::usb_handler @ src/main.rs:306
9 INFO  USB interrupt received.
└─ blinky_nucleo_h743zi2_interrupt::app::usb_handler @ src/main.rs:306
10 INFO  USB interrupt received.
└─ blinky_nucleo_h743zi2_interrupt::app::usb_handler @ src/main.rs:306
11 INFO  USB interrupt received.
└─ blinky_nucleo_h743zi2_interrupt::app::usb_handler @ src/main.rs:306
12 INFO  USB interrupt received.
└─ blinky_nucleo_h743zi2_interrupt::app::usb_handler @ src/main.rs:306
13 INFO  USB interrupt received.
└─ blinky_nucleo_h743zi2_interrupt::app::usb_handler @ src/main.rs:306
14 INFO  USB interrupt received.
└─ blinky_nucleo_h743zi2_interrupt::app::usb_handler @ src/main.rs:306
15 INFO  USB interrupt received.
└─ blinky_nucleo_h743zi2_interrupt::app::usb_handler @ src/main.rs:306
16 INFO  USB interrupt received.
└─ blinky_nucleo_h743zi2_interrupt::app::usb_handler @ src/main.rs:306
17 INFO  USB interrupt received.
└─ blinky_nucleo_h743zi2_interrupt::app::usb_handler @ src/main.rs:306

Disasm, 2021-06-24:
Thank you! I'll try to look into this when I have time. This looks really strange as in your chip these two USB peripherals should be identical.

TomDeRybel, 2021-06-24:
Thank you for looking into this! Please let me know if I can do anything else to help.
(I suspect that the OTG1_HS identifying as a low-speed device has to do with the parasitic bus loading,)

hacknus, 2023-09-08:
As I already described in [this issue](https://github.com/stm32-rs/stm32f4xx-hal/issues/544#issuecomment-1711704863) I noticed that when connecting a stm32f405 to a linux (ubuntu) host, the interrupts dont get triggered but polling in main works fine. 
When connecting to a windows or macos host, the interrupts get called fine.

Dirbaio, 2024-04-12:
can confirm, f429 disco, also interrupts don't work.

I've found it's somehow related to WFE/WFI. 
- The stm32f4xxh-hal example works, but fails once you add a WFE [here](https://github.com/stm32-rs/stm32f4xx-hal/blob/master/examples/usb-serial-irq.rs#L69).
- embassy-stm32 (separate implementation, doesn't use this crate)works if you remove the `wfe` in the executor, fails if you leave it in.

so it seems like some sort of hardware errata... but there's nothing in the errata sheet! :(

Dirbaio, 2024-04-12:
found it. you have to set `ahb1lpenr.usb_otg_hs_ulpilpen=0`. For some reason that defaults to 1, and it kicks in when entering WFE/WFI, sabotaging the phy choice...

https://github.com/embassy-rs/embassy/pull/2805