stm32-rs/synopsys-usb-otg #18: stm32f407 discovery serial example does not work

niclashoyer, 2021-04-12:
Hi,

I tried to run the serial example on my stm32f407 discovery board: https://github.com/Disasm/usb-otg-workspace/blob/master/example-f407-board/examples/serial.rs

The code runs fine so far, the device enumerates:

```bash
$ lsusb
Bus 004 Device 008: ID 0483:3748 STMicroelectronics ST-LINK/V2
Bus 002 Device 012: ID 16c0:27dd Van Ooijen Technische Informatica CDC-ACM class devices (modems)
…
```

But it does not send/receive any characters. I already tried to investigate it, as there is no error so far. The only thing that I see is that it never actually receives data. I already tried several clients (screen, tio, minicom), to no avail. I also uninstalled ModemManager, just in case.

Using `usbmon` and wireshark I see that the device connects successfully, but then just stops communicating after the `SET LINE CODING Request`:

[image]

Full wireshark dump: [usb.pcapng.gz](https://github.com/stm32-rs/synopsys-usb-otg/files/6297342/usb.pcapng.gz)

Is there any way I could debug this further?

niclashoyer, 2021-04-12:
Just tested on a different computer and it worked there, strange. `dmesg` output was also a bit different. I'll try to dump usb messages there and compare them.

Disasm, 2021-04-12:
A silly question: a you running a release build of a firmware?

niclashoyer, 2021-04-12:
yes, the firmware was identical (actually flashed it on the non-working computer and just plugged it into the other computer)

niclashoyer, 2021-04-12:
Ok, here is the dump of the communication from the working computer:

[usb.pcapng.gz](https://github.com/stm32-rs/synopsys-usb-otg/files/6298373/usb.pcapng.gz)

As far as I can see the communication is the same until `URB_CONTROL out` and then it just starts communicating (I just pressed `abc`):

[image]

I already tested on another computer and it also worked there, so I think it must be something with the OS, but I don't know how to proceed from here.

I think this can be closed for now.

closed 2021-04-12 by niclashoyer