stm32-rs/synopsys-usb-otg #21: The USB set address request failed in debug build

zgdump, 2022-01-02:
Hi!

I used usb cdc in my project for STM32F411CEU and noticed that sometimes Windows shows error 43 in Device Manager. If reconnect device, then everything works as it should. To reproduce the issue, I plugged and popped the device several times

I found that the problem is reproduced less frequently if the frequency of sysclk is increased. And it completely goes away build project in release mode

<details>
  <summary>Code:</summary>

  Just poll and read, nothing more

  ```rust
  static mut EP_MEMORY: [u32; 1024] = [0; 1024];
  
  #[entry]
  fn main() -> ! {
      rtt_init_print!();
      rprintln!("Hello!");
  
      let dp = pac::Peripherals::take().unwrap();
  
      let rcc = dp.RCC.constrain();
  
      let clocks = rcc
          .cfgr
          .use_hse(25.mhz())
          .sysclk(48.mhz())
          .require_pll48clk()
          .freeze();
  
      let gpioa = dp.GPIOA.split();
      let usb = USB {
          usb_global: dp.OTG_FS_GLOBAL,
          usb_device: dp.OTG_FS_DEVICE,
          usb_pwrclk: dp.OTG_FS_PWRCLK,
          pin_dm: gpioa.pa11.into_alternate(),
          pin_dp: gpioa.pa12.into_alternate(),
          hclk: clocks.hclk(),
      };
  
      let mut usb_bus = UsbBus::new(usb, unsafe { &mut EP_MEMORY });
  
      let mut serial = SerialPort::new(&usb_bus);
  
      let mut usb_dev = UsbDeviceBuilder::new(&usb_bus, UsbVidPid(0x16c0, 0x27dd))
          .product("Serial port")
          .device_class(USB_CLASS_CDC)
          .build();
  
      loop {
          if !usb_dev.poll(&mut [&mut serial]) {
              continue;
          }
  
          let mut buf = [0u8; 64];
  
          match serial.read(&mut buf[..]) {
              Ok(count) => {}
              Err(UsbError::WouldBlock) => {}
              Err(err) => {}
          };
  
          match serial.write(&[0x3a, 0x29]) {
              Ok(count) => {}
              Err(UsbError::WouldBlock) => {}
              Err(err) => {}
          };
      }
  }
  ```
</details>

I was unable to get dump using wireshark, but I recorded what was happening on usb using a logic analyzer:

[Rust Usb Bug LA Dump.zip](https://github.com/stm32-rs/synopsys-usb-otg/files/7799711/RUST.USB.CDC.BUG.zip)

I use a ready-made black pill board from the Chinese

[image]

The problem is not reproducible on a project written in C using CubeMX

zgdump, 2022-01-02:
My friend sent me such a link when I described my problem to him: https://community.st.com/s/question/0D50X0000B8jzZL/why-does-the-stm32f407-send-a-stall-on-usbfs

ajxville, 2023-09-11:
Is there any hope of getting this remedied?

ajxville, 2024-04-29:
Just checking in on this again. Any hope for a fix?